cedricp/ddt4all #1155: can't write certain options

bigmac5753, 2024-07-06:
how can I write to these options?

(I know it's in demo mode, I get the same in connected mode)

[image]

AndyIreland, 2024-07-06:
hi, its because that ecu parameter can only be read, not modified - there are no boxes on the 'write' for you to put your parameters into. 
Whether you can use the manual command 

[image]

is another matter, you might be able to, but its a risk.

bigmac5753, 2024-07-07:
I found the options to read, but not write, I guess it can't be done.

[image]
[image]

AndyIreland, 2024-07-07:
I think thats the case