cedricp/ddt4all #1175: How can I generate or find a file for a module for my car?

vlovini, 2024-07-24:
Hi colleagues!

First of all, congratulations on this impressive project. I will see if I can contribute by my self in the future :)

I just bought a Sandero RS (Sandero II in its brazilian hot hatch version) and want to activate some things that I'm missing from my previous car (I came from a Ford, so I'm used to the Forscan softwares and modules).

How can I find or generate configuration files for some modules of my car? Being more specifically, I need a USM External lighting module that didn't match with any of the ones available in the ecu.zip that I found on the internet.

Is it possible to extract this from the car or from a maintenance site?

Thanks in advance!

vlovini, 2024-07-24:
Found out there is a place for questions. Sorry, moving to another discussion place.

SimonBard, 2024-08-26:
@vlovini: Can you please link the place you mentioned? Or better directly tell me where to find the database?