cedricp/ddt4all #1201: Win11x64 PermissionError ELM 327 USB

MisterAssm, 2024-08-21:
Bonjour,
Je bloque depuis un petit moment sur l'application, lorsque j'appuie sur Tests ELM, j'ai une PermissionError (malgré le fait que j'essaye d'ouvrir DDT en administrateur) ou bien OSError parce qu'il ne trouve pas l'ELM.
L'ELM est pourtant fonctionnel avec d'autres applications (EOBD). J'ai essayé plusieurs ports.

Si quelqu'un aurait une idée d'où pourrait venir l'erreur..

[image]

Furtif, 2024-08-21:
vous êtes sur que c'est le port com OBD ??? Ce error arrive si le port est en use ....

MisterAssm, 2024-08-21:
> vous êtes sur que c'est le port com OBD ??? Ce error arrive si le port est en use .... 

Je pense, du moins le port fonctionne sans problème avec d'autres applications (EOBD). J'avais essayé dans le doute avec les deux ports de mon pc mais même problème.

Furtif, 2024-08-22:
> > vous êtes sur que c'est le port com OBD ??? Ce error arrive si le port est en use ....
> 
> Je pense, du moins le port fonctionne sans problème avec d'autres applications (EOBD). J'avais essayé dans le doute avec les deux ports de mon pc mais même problème.

oui possible, mais ddt4all ne fonctionne pas avec tous OBD il faudra utiliser l'un de ceux mentionné dans readme https://github.com/cedricp/ddt4all?tab=readme-ov-file#supported-diagnostic-adapters-so-far

j'ai 2 OBD un wifi l'autre USB et les 2 fonctionnent bien sous win11.

Furtif, 2024-08-22:
> > > > vous êtes sur que c'est le port com OBD ??? Ce error arrive si le port est en use ....
> > > 
> > > 
> > > Je pense, du moins le port fonctionne sans problème avec d'autres applications (EOBD). J'avais essayé dans le doute avec les deux ports de mon pc mais même problème.
> > 
> > 
> > oui possible, mais ddt4all ne fonctionne pas avec tous OBD il faudra utiliser l'un de ceux mentionné dans readme https://github.com/cedricp/ddt4all?tab=readme-ov-file#supported-diagnostic-adapters-so-far
> > j'ai 2 OBD un wifi l'autre USB et les 2 fonctionnent bien sous win11.
> 
> En voulant vérifier, la page sur laquelle j'ai commandé mon elm n'est plus disponible.. J'en recommande un autre, c'est peut être ça le problème. Je vous tiens au courant.

OK ok 👍

Furtif, 2024-08-22:
> vous êtes sur que c'est le port com OBD ??? Ce error arrive si le port est en use ....
>
>
> Je pense, du moins le port fonctionne sans problème avec d'autres applications (EOBD). J'avais essayé dans le [...]

Par contre soit sur qu'il a un chip PIC18F25K80 comme décrit dans readme !!!!

Furtif, 2024-08-22:
[ODB PIC18F25K80](https://www.amazon.fr/Diagnostic-Voiture-Danalyse-PIC18F25K80-Protocoles/dp/B0C6H5HD2T/ref=sr_1_1_sspa?dib=eyJ2IjoiMSJ9.QxuazZCAtri4JIqhZC-ePPkRT1gu98Q2d0skkP4DS2IOuBF2yWxfG_9TIq5XN5hOqFGDTKgcOF45kMdQ6iLdOjJQmFL4KZA_0sojLAff_6lVPFUjNu7GNF9VEukhZ-bTyf5QpQRzua8eq3uLnPyv47ZmSA7E4lZQdStzjp-kZT5AmWxD4Tn7xkFaEHo3NKpncdPjho26IN-K6GRYgqbzjL5MHb7PaLXpETXAPc8rU66iBCznhVEc62YwX9maBN-DB43Y7yrEjn9xHVg-FIf9ud7gx_2yNZqBdXG4BZzZqJY.DkGxYlMm-Wk1GmpuLyP44NaFVJcJTcgbRoSk4T1VE4k&dib_tag=se&keywords=pic18f25k80&qid=1724337857&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1) 

type celui la.

Furtif, 2024-08-22:
Ou sa https://github.com/cedricp/ddt4all/discussions/1205#discussioncomment-10421650

Furtif, 2024-08-22:
vais changer ceci vers discussions car il n'est pas tout a fait un issue, si était le cas il aurais plusieurs users avec le même problème.