cedricp/ddt4all #1202: Help me please

dee115, 2024-08-22:
Hello, I went to this window and clicked on the command key off / key on restart and the screen became completely black. Can anyone help me please?

[image]
[image]

AndyIreland, 2024-08-22:
are you sure it was not the 'soft reset' button you pressed instead? 

try press the 'Reset Hard' button there and see if rlink2 re-boots OK and screen comes back - at this stage it shouldnt hurt.

dee115, 2024-08-22:
Excuse me, where is this button you mean?

dee115, 2024-08-22:
> are you sure it was not the 'soft reset' button you pressed instead?
> 
> try press the 'Reset Hard' button there and see if rlink2 re-boots OK and screen comes back - at this stage it shouldnt hurt.

Excuse me, where is this button you mean?

AndyIreland, 2024-08-22:
> > are you sure it was not the 'soft reset' button you pressed instead?
> > try press the 'Reset Hard' button there and see if rlink2 re-boots OK and screen comes back - at this stage it shouldnt hurt.
> 
> Excuse me, where is this button you mean?

[image]