cedricp/ddt4all #1204: DDT4All on macOS M1: Severe Lag When Editing ECUs After Scanning

enesokyay, 2024-08-22:
Hello,

I am using DDT4All on a MacBook with an M1 processor. The installation went smoothly, and the software runs without any initial issues. However, I am experiencing significant performance problems when attempting to edit any ECU after scanning.

Steps to Reproduce:

	1.	I connected my ELM327 OBD-II adapter (as shown in the attached image).
	2.	I launched DDT4All and successfully scanned all ECUs in my Renault Clio IV.
	3.	The scanning process works without any problems.
	4.	However, when I attempt to select and edit any ECU from the list, the application becomes extremely slow and unresponsive. The delay is so severe that I am unable to perform any meaningful operations.

Screenshots:

	•	I have attached screenshots of the application during the scanning process and the list of ECUs.
	•	Additionally, I have included a photo of the OBD-II adapter I am using.

System Information:

	•	MacBook Model: MacBook Pro with M1 Chip
	•	Operating System: macOS Ventura (or specify your version if different)
	•	DDT4All Version: 3.0.1 (Stable)

Troubleshooting Steps Already Taken:

	•	I have ensured that the software is up-to-date.
	•	I have tried restarting the application and the MacBook, but the issue persists.
	•	I have also tried using a different OBD-II adapter, but the problem remains the same.

Questions:

	•	Is this a known issue with the M1 processor, or could it be related to the compatibility of the software with macOS?
	•	Are there any suggested workarounds or settings that could improve performance?
	•	Would running DDT4All in a virtual machine (such as Parallels or VirtualBox) with Windows improve performance?

Thank you for your assistance!

[image] [image] [image]

enesokyay, 2024-08-22:
I suspect that the issue might be related to the OBD adapter, so I have ordered a new one from AliExpress

6,64€ | ELM327 V2.1 OBD2 Tarayıcı TR6260S1 BT/Wifi ELM 327 Android /IOS PK Vgate Icar2 Kod Okuyucu Için OBD Araç Teşhis Aracı https://a.aliexpress.com/_EzASCjj

Furtif, 2024-08-22:
for edit data fields you need unlock stuff 
[image]

enesokyay, 2024-08-22:
> Veri alanlarını düzenlemek için bir şeylerin kilidini açmanız gerekir ![görüntü](https://private-user-images.githubusercontent.com/11375035/360449883-0ca1171c-559a-47d8-aa73-39eb2da35346.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzMzQwMDMsIm5iZiI6MTcyNDMzMzcwMywicGF0aCI6Ii8xMTM3NTAzNS8zNjA0NDk4ODMtMGNhMTE3MWMtNTU5YS00N2Q4LWFhNzMtMzllYjJkYTM1MzQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIyVDEzMzUwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjN2E3NzZhNGYwOGRlNThlNzJkYjk5M2E1MTBmMjJjMzc5Yzc5YmU3NmU4YmQ0M2U1MGE1MjQ5YzU3YjUwMzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uWXUW0MuHKHVNEVaADCXueuq4mZw7A1Bz6-wksOtclg)

Don’t I need to first go to the editing screen to activate it? Because after scanning the ECUs with CAN, the entire system freezes. My computer takes almost a minute to respond to any clicks. From what I’ve seen in all the videos, it seems I need to press that button to make any corrections.

Furtif, 2024-08-22:
> Je soupçonne que le problème pourrait être lié à l'adaptateur OBD, j'en ai donc commandé un nouveau sur AliExpress
> 
> 6,64€ | ELM327 V2.1 OBD2 Tarayıcı TR6260S1 BT/Wifi ELM 327 Android /IOS PK Vgate Icar2 Kod Okuyucu ici OBD Araç Teşhis Aracı https://a.aliexpress.com/_EzASCjj

c'est possible aussi, mais votre image de test ecus a l'air de bien fonctionner.

Furtif, 2024-08-22:
> > Veri alanlarını düzenlemek için bir şeylerin kilidini açmanız gerekir![vue d'ensemble](https://private-user-images.githubusercontent.com/11375035/360449883-0ca1171c-559a-47d8-aa73-39eb2da35346.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzMzQwMDMsIm5iZiI6MTcyNDMzMzcwMywicGF0aCI6Ii8xMTM3NTAzNS8zNjA0NDk4ODMtMGNhMTE3MWMtNTU5YS00N2Q4LWFhNzMtMzllYjJkYTM1MzQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIyVDEzMzUwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjN2E3NzZhNGYwOGRlNThlNzJkYjk5M2E1MTBmMjJjMzc5Yzc5YmU3NmU4YmQ0M2U1MGE1MjQ5YzU3YjUwMzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uWXUW0MuHKHVNEVaADCXueuq4mZw7A1Bz6-wksOtclg)
> 
> Ne dois-je pas d'abord aller sur l'écran d'édition pour l'activer ? Car après avoir scanné les ECU avec CAN, tout le système se bloque. Mon ordinateur met presque une minute à répondre aux clics. D'après ce que j'ai vu dans toutes les vidéos, il semble que je doive appuyer sur ce bouton pour effectuer des corrections.

quand vous cliquez sur ```write``` sur n'importe quel setting a écrire il montre un message warning, pour debloquer faudra cliquer la.

enesokyay, 2024-08-22:
> c'est possible aussi, mais votre image de test ecus a l'air de bien fonctionner.

Oui, c’est justement cela qui est étrange. Je pense qu’il y a un problème de vitesse avec l’appareil, c’est pourquoi j’en ai commandé un nouveau. Cependant, je n’ai aucun problème tant que je ne sélectionne pas un ECU dans la section du module de contrôle de la carrosserie. Dès que je sélectionne un ECU, l’ordinateur ralentit instantanément.

enesokyay, 2024-08-22:
> > > Veri alanlarını düzenlemek için bir şeyin kilidini açmanız gerekir![vue d'ensemble](https://private-user-images.githubusercontent.com/11375035/360449883-0ca1171c-559a-47d8-aa73-39eb2da35346.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzMzQwMDMsIm5iZiI6MTcyNDMzMzcwMywicGF0aCI6Ii8xMTM3NTAzNS8zNjA0NDk4ODMtMGNhMTE3MWMtNTU5YS00N2Q4LWFhNzMtMzllYjJkYTM1MzQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIyVDEzMzUwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFjN2E3NzZhNGYwOGRlNThlNzJkYjk5M2E1MTBmMjJjMzc5Yzc5YmU3NmU4YmQ0M2U1MGE1MjQ5YzU3YjUwMzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uWXUW0MuHKHVNEVaADCXueuq4mZw7A1Bz6-wksOtclg)
> > 
> >  Ne dois-je pas d'abord aller sur l'écran d'édition pour l'activer ? Car après avoir scanné les ECU avec CAN, tout le système se bloque. Mon ordinateur met presque une minute à répondre aux clics. D'après ce que j'ai vu dans toutes les vidéos, il semble que je doive appuyer sur ce bouton pour effectuer des corrections.
> 
> quand vous cliquez sur `write` sur n'importe quel setting a écrire il montre un message warning, pour debloquer faudra cliquer la.

C’est exactement ça le problème. Je ne peux pas accéder à l’écran de configuration pour écrire

Furtif, 2024-08-22:
> > > Je soupçonne que le problème pourrait être lié à l'adaptateur OBD, j'en ai donc commandé un nouveau sur AliExpress 
> > > 6,64€ | ELM327 V2.1 OBD2 Tarayıcı TR6260S1 BT/Wifi ELM 327 Android /IOS PK Vgate Icar2 Kod Okyucu ici OBD Araç Teşhis Aracı https://a.aliexpress.com/_EzASCjj
> > 
> > 
> > c'est possible aussi, mais votre image de test ecus a l'air de bien fonctionner.
> 
> Oui, c'est justement cela qui est étrange. Je pense qu'il y a un problème de vitesse avec l'appareil, c'est pourquoi j'en ai commandé un nouveau. Cependant, je n'ai aucun problème tant que je ne sélectionne pas un ECU dans la section du module de contrôle de la carrosserie. Dès que je sélectionne un ECU, l'ordinateur ralentit instantanément.

double-click sur l'ecu que vous voulais puis section ecrans plus bas.

Furtif, 2024-08-22:
car ici https://private-user-images.githubusercontent.com/178347757/360447738-f4c6c302-4619-4368-8988-1df9c38880fc.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzMzQzNTAsIm5iZiI6MTcyNDMzNDA1MCwicGF0aCI6Ii8xNzgzNDc3NTcvMzYwNDQ3NzM4LWY0YzZjMzAyLTQ2MTktNDM2OC04OTg4LTFkZjljMzg4ODBmYy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MjJUMTM0MDUwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjE2ZWRiOTU3NjVkZWQwNjJjNTZiOGZlODU3NWI0MDkxOTYyODczMzM2YjdkOTNmMzk5MzhlN2FhZGFiZTUwMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.upVjeuHfcY2_h3f9G-dSdqk0CJs-7CsDjztgutwRThk ne ne vois pas ligne dans section ecrans

Furtif, 2024-08-22:
double-click la 
[image]
apres il aura dans la section ecrans les stuff pour edit.

Furtif, 2024-08-22:
comme sa double-clik sur l'ecu 
[image]

enesokyay, 2024-08-22:
Je n'arrive pas à expliquer le problème que j'ai. J'ai fait une vidéo mais je ne peux pas la télécharger ici.

Furtif, 2024-08-22:
> Je n'arrive pas à expliquer le problème que j'ai. J'ai fait une vidéo mais je ne peux pas la télécharger ici.

oky

enesokyay, 2024-08-22:
Lorsque j’arrive à l’écran qui indique ‘FENETRE ECRAN’, le programme ne fonctionne pas. Je ne peux effectuer aucune action. Cela prend environ une minute pour réagir.

Furtif, 2024-08-22:
> Lorsque j’arrive à l’écran qui indique ‘FENETRE ECRAN’, le programme ne fonctionne pas. Je ne peux effectuer aucune action. Cela prend environ une minute pour réagir.

avais vous une BD bien ?

enesokyay, 2024-08-22:
Je rentre chez moi et je vous ferai parvenir la vidéo.

Furtif, 2024-08-22:
si c'est pas le cas https://github.com/cedricp/ddt4all/discussions/845 prennez celle-ci (ne dezipez pas placez le zip dans root de votre source python)

Furtif, 2024-08-22:
> trim.CDE46854-5ECE-4A7F-8F6F-43EB06E39E8B.MOV

vous parlais du délais a load list des écrans config ?, oui sa c'est normal un délai, il y a 4000 et quelque ecus e des milliers ecrans, mise a part une DB custom (selemente avec le vehicule) ou un ordinateur de la nasa peut etre plus rapide.

enesokyay, 2024-08-22:
Le problème est que ces bégaiements ne disparaissent jamais. Je sais que j'ai attendu 30 minutes pour le premier scan de l'ecu. ensuite j'ai été patient et j'ai attendu à nouveau pour le faire de cette façon, mais je n'obtiens aucun résultat. si vous remarquez dans la vidéo, il n'y a aucun changement même si je bouge la souris. j'ai regardé des vidéos sur youtub et les gens attendent juste le premier démarrage.

Furtif, 2024-08-22:
> Le problème est que ces bégaiements ne disparaissent jamais. Je sais que j'ai attendu 30 minutes pour le premier scan de l'ecu. ensuite j'ai été patient et j'ai attendu à nouveau pour le faire de [...]

oui vu sa tester sur mon macOs j'ai pas autant délai .... chelou!

quand vous cliquez sur > pour deployer la section sa rame aussi ?

enesokyay, 2024-08-22:
j'ai pu accéder à l'écran une fois (après environ 35 minutes). j'ai fait le même processus en regardant une vidéo sur youtub mais je n'ai pas pu obtenir de données parce que j'ai sélectionné le mauvais ecu dans mon véhicule. dois-je attendre 45 minutes pour chaque ecu afin de sélectionner le bon ecu ? ce processus ne devrait pas prendre autant de temps.

et Oui, il se bloque ainsi lorsque je sélectionne quoi que ce soit sur l'écran `ecu windows`.

Furtif, 2024-08-22:
en mode Démo il y a le même problème? (j'ai remarquer que vous êtes en mode connecté, peut être sa prends du temps pendant que read les infos du ecu)

Furtif, 2024-08-22:
Si en mode Démo rien de cella se passe, peut-être le problème est lie au périphérique OBD quand il tente de lire les informations du Ecu

enesokyay, 2024-08-22:
Je n'ai aucun problème en mode démo.

Oui, j'ai pris une vidéo pendant que j'étais connecté à l'appareil. mais peu importe le temps que j'attends, ça ne s'améliore pas. je pensais que ça prenait du temps à charger. j'ai scanné l'ecu avec can pendant 35 minutes. puis j'ai attendu 30 minutes pour 1 écran d'ecu (malgré le scan) et j'ai vu que ce n'était pas le bon ecu. je n'ai même pas essayé d'essayer pour le deuxième, j'ai commandé un obd à nouveau et j'ai commencé un fil de discussion sur le forum.

Furtif, 2024-08-22:
vais convertir ce issu en discussion,  si mode Démo aucun bug alors c'est le OBD que est mauvais.

enesokyay, 2024-08-22:
J'y ai pensé aussi, mais à ce stade, c'est bizarre qu'il scanne tous les écus sur l'écran du terminal.