cedricp/ddt4all #1213: Elm327 wifi problème avec mac m1

enesokyay, 2024-08-24:
Issue: Connection to ELM327 Wi-Fi Device Fails - Stuck on “Connecting to port 35000”

Description:

I am trying to connect DDT4All to my ELM327 Wi-Fi device on my MacBook running macOS. The device is connected to the Wi-Fi network, and I can confirm that the port 35000 is open and accessible (as seen in the attached terminal output). However, when I run DDT4All, it repeatedly tries to connect to the port but fails to establish a connection.

Steps Taken:

	1.	Wi-Fi Settings:
	•	Manually set the IP address to 192.168.0.2.
	•	Subnet mask set to 255.255.255.0.
	•	Router (Gateway) set to 192.168.0.10.
	•	Disabled IPv6.
(Please see the attached screenshot for my Wi-Fi configuration.)
	2.	Port Check:
	•	Using the nc -zv 192.168.0.10 35000 command, I confirmed that the port 35000 on the ELM327 device is open and reachable.
	3.	DDT4All Configuration:
	•	I verified that the IP and port in the DDT4All source code are correctly set to 192.168.0.10:35000.
(Please refer to the attached screenshots of the source code.)
	4.	Issue:
	•	Despite all these settings, DDT4All continues to try to connect to the ELM327 device, but the connection is never established. The software remains stuck on “Connecting to port 35000,” as shown in the terminal output screenshot.

Screenshots:

	•	Wi-Fi Configuration: [Insert attached screenshot of Wi-Fi settings]
	•	Port Check: [Insert attached screenshot of terminal output showing successful port check]
	•	DDT4All Terminal Output: [Insert attached screenshot of terminal output showing repeated connection attempts]
	•	Source Code: [Insert attached screenshot of source code showing IP and port settings]

Additional Information:

	•	The ELM327 device works fine when connected to a mobile device (phone) using the same Wi-Fi settings, so the issue seems to be specific to the MacBook or the DDT4All configuration on macOS.

Request:

Any guidance on what might be causing this issue or how to troubleshoot further would be greatly appreciated. Is there something specific to macOS or the DDT4All setup that could be causing this connection failure?

[image]
[image]
[image]
[image]

Furtif, 2024-08-24:
as tu tester sa vraiment connecté dans le véhicule ?

enesokyay, 2024-08-24:
Yesterday I made a connection with my phone and I was able to make a change, but the application on the phone is quite limited. I can enter from my iPhone 12 device, both hand and the car work, but . I can't connect when I switch to the Mac

enesokyay, 2024-08-24:
[image]

I come across this when I try this way.

Furtif, 2024-08-25:
je crois que tas pas laisser le temps a github pour upload l'image...

enesokyay, 2024-08-25:
> je crois que tas pas laisser le temps a github pour upload l'image...
> 
> https://github.com/cedricp/ddt4all/issues/1213#issuecomment-2308451041
> 
> stressé probablement...

🫡🫡

enesokyay, 2024-08-25:
> je crois que tas pas laisser le temps a github pour upload l'image...
> 
> https://github.com/cedricp/ddt4all/issues/1213#issuecomment-2308451041
> 
> stressé probablement...

I've been trying since last night I still haven't solved the problem. I even installed a Windows for Mac...

Furtif, 2024-08-25:
> > je crois que tas pas laisser le temps a github pour upload l'image...
> > [#1213 (comment)](https://github.com/cedricp/ddt4all/issues/1213#issuecomment-2308451041)
> > stressé probablement...
> 
> I've been trying since last night I still haven't solved the problem. I even installed a Windows for Mac...

worked ??

enesokyay, 2024-08-25:
> worked ??

Non 😑[image][image][image]

enesokyay, 2024-08-25:
I can enter my obd device with my iPhone phone.[image]

Furtif, 2024-08-25:
Bizarre tout sa, j’ai aussi se app dans mon iPhone mais mon obd2 Wi-Fi fonctionne grave bien avec tout ceci.

Furtif, 2024-08-25:
Vous aviez cliquer sur l’icone Wi-Fi dans ddt4all ?

enesokyay, 2024-08-25:
> Vous aviez cliquer sur l’icone Wi-Fi dans ddt4all ?

Oui

enesokyay, 2024-08-25:
> Bizarre tout sa, j’ai aussi se app dans mon iPhone mais mon obd2 Wi-Fi fonctionne grave bien avec tout ceci.
> 
> 
> 
> 

Il n'y a pas de problème avec mon téléphone. Mais je ne peux pas me connecter avec mac. Je pense que je vais devoir acheter un obd connecté par USB...

Furtif, 2024-08-25:
> > Bizarre tout sa, j’ai aussi se app dans mon iPhone mais mon obd2 Wi-Fi fonctionne grave bien avec tout ceci.
> 
> Il n'y a pas de problème avec mon téléphone. Mais je ne peux pas me connecter avec mac. Je pense que je vais devoir acheter un obd connecté par USB...

peut être désactivant le firewall, mais bon j'ai pas eu besoin de faire sa dans mon win11 ou mon macos ou linux debian.

enesokyay, 2024-08-25:
> peut être désactivant le firewall, mais bon j'ai pas eu besoin de faire sa dans mon win11 ou mon macos ou linux debian.

Cela m'est venu à l'esprit et je l'ai fait, mais le résultat n'a pas changé.