cedricp/ddt4all #1226: Scanning not working

Sxc808, 2024-08-31:
When I press the scan/magnifying glass to scan  none of the green or red text show up on the bottom and it just says “scanning can” a whole bunch of codes come up on the python but nothing on the ddt4all, I have downloaded the 2drive software with all cars database can someone please help

AndyIreland, 2024-08-31:
what year is your car , it may be protected with security gateway firewall on the OBD2 socket if its one of a late model. 
Have you got the correct ELM327 with the required PIC chip on the board or are you using a cheap chinese clone?

Sxc808, 2024-08-31:
My Clio rs 200 is a 2014, yes my obd2 is the usb one with the pic as well as the switch for HS-can and MS-can

AndyIreland, 2024-08-31:
> My Clio rs 200 is a 2014, yes my obd2 is the usb one with the pic as well as the switch for HS-can and MS-can

Ah right, I find those ones with the switch on them are not wired up correctly - they seem to be wired up for FORSCAN (Ford HS/MS?) and I shouldnt think with a Pic18F25K80 chip on them 

what ecu's are you trying to read? - the rlink/medianav radio on CAN2 / CAN-HI  ? - or all other ECU's on the car?

Sxc808, 2024-08-31:
My r link system with rs monitor

AndyIreland, 2024-08-31:
> My r link system with rs monitor

ya wired up wrong then i would say for a start - FORSCAN with a chrome metal toggle switch on the case and wired up wrong to read rlink1 on the CAN2 / CAN-H through the OBD2 socket. 
Follow the wiring diagram on here - it says for adding AA to rlink but its the same thing, if you want to do any reading writing on the rlink through the OBD2 socket you have to wire up either a y-cable like this or re-solder the elm327 adaptor to read /write rlink.
https://www.gps-rlink.com/r-link-1/activate-android-auto-waze-on-r-link-1/