cedricp/ddt4all #1243: Ambient lights and functions

Kiritakas, 2024-09-11:
I'll be honest, I'm new to ddt4all, and I need advice first :D.
 I own Talisman initiale paris 2016.
 
the short story goes: 

previous owner(family) bricked the system, by going into developer settings and pressing mid display setting.
someone tried to reinstall whole system ( the mechanic claimed he bought a script, and so on )  . and it fixed the black screen, but
It destroyed everything else. To use the BT with iphone i had to download a song from itunes, just so i could select the BT multimedia setting and play the songs that i liked :D.

Problem with ambient lights, is really strange one.
ddt4all 
Found ECU [Audio] (not perfect match) :RadNav 1,4 to 2.x DIAGVERSION [OD] SUPPLIER, [CAP] SOFT [33) VERSION [16986 instead 16966)
 ECU config III - ambient lights are present and activation requests and etc.
 ECU config II - The HMI settings dd4all  showed everything as ( not present ), BUT developer settings showed ambient light area 1/2/3 present/ colors present. ( maybe because of version mismatch? )

Also when I try to change the settings of ambient lights,  it reverts itself back. excluding colors. screen and the speedometer changes the color and it stays that way. but when i try to turn on the ambient lights, it stays on for couple of seconds and reverts itself back to off.
To clarify ( lights stay off, when turning the setting on )
 
 + I refer to that debug mode when saying developer mode :D.
if it's something stupid please don't be harsh. xd I'm learning

Kiritakas, 2024-09-11:
whops..... i created it, in issues tab sorry about that 😅😅😅

Furtif, 2024-09-13:
https://github.com/cedricp/ddt4all/discussions/1240 read this

closed 2024-09-13 by Furtif