cedricp/ddt4all #1249: Megane 4 high and low beam flashing at the same time

Deswa16, 2024-09-17:
Can you show me a setting that I can make to make the Megane 4 high and low beams flash at the same time?

AndyIreland, 2024-09-17:
I do believe from memory on my Clio Mk4 that either headlamp bi function or Dual bulbs on the EMM Lighting ECU was the setting for that (Change from 0 to 1)  - but I might be wrong. 

[image]

Deswa16, 2024-09-17:
> Clio Mk4'ümde hafızamdan bildiğim kadarıyla bunun için EMM Aydınlatma ECU'sunda far çift fonksiyonu veya Çift ampul ayarının olması gerekiyordu (0'dan 1'e değiştir) - ama yanılıyor da olabilirim.
> 
> ![görüntü](https://private-user-images.githubusercontent.com/49172217/368159335-d6c34973-34a2-4e40-8de9-0d755fc42da3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY1Nzc1NzAsIm5iZiI6MTcyNjU3NzI3MCwicGF0aCI6Ii80OTE3MjIxNy8zNjgxNTkzMzUtZDZjMzQ5NzMtMzRhMi00ZTQwLThkZTktMGQ3NTVmYzQyZGEzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE3VDEyNDc1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmZmNkNDUxY2I2OWMzYWRhOTQ3NGE1YzIwNWMyNjdmODI0NzgyMWQwYWRlNmVmODdmMzQ1MzU2YTc4MWRjNjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.y-gaKMeXoPnafesGkJGWLyQP0301FlkKfSWrxpnxZ04)

This setting does not appear in the Megane 4 USM brain