cedricp/ddt4all #1257: Ddt4all reset go wrong R link2 2018 Megane

Endkod, 2024-09-24:
Can someone help me with this problem.
No sound/camera/temperature control after ddt4all reset. The screen was also black, which I resolved with a macro, which can still be done to get it working again.

AlMeu, 2024-09-24:
> Can someone help me with this problem. No sound/camera/temperature control after ddt4all reset. The screen was also black, which I resolved with a macro, which can still be done to get it working again.

What software version you have in your unit?
A photo will help to understand your situation.

Endkod, 2024-09-24:
[image]
Before

AlMeu, 2024-09-24:
The upgrade from v2.2 or v3.3 to higher version is done only in Renault dealer services or by someone who knows very well what and how to do.
I messed up the software upgrade wrong.
Go to a Renault service maybe they will get fixed.

Endkod, 2024-09-24:
I think I reset the radio dtc because I had a fault in my tire sensor. Renault quickly checked with the can clip but could not update the radio, that was on my holiday in France. They said error in radio and automatic transmission. Megane DCI gt was delivered with software 7. and is from 2018. I think I need a macro to set everything right.

Endkod, 2024-09-24:
R link 2 screen works just shows radio time... but no sound, no multi sense, camera

Endkod, 2024-09-24:
I have tried to perform an radio update myself with Canclip but it stops at 55% error automatic gearbox please contact information......

AlMeu, 2024-09-24:
Following your unit behavior as you describe it is looking like an upgrade from v3.3 to v9.0 wrong.
If the CanClip refuse the upgrade it is end of story.
Just remains to change the unit with another new unlocked.

Endkod, 2024-09-24:
Then I made a mistake with Ddt4all, I reset the radio to DTC. R link software 9. was upgraded last year from 7 to 9. It just worked until I reset it with Ddt4all 2 weeks ago

matzel82, 2024-09-24:
Disconnect Clip and Battery and wait a time at 1h.
Than reconnect. Try Update again.

AlMeu, 2024-09-24:
If you say so, that it is.
But the fact it is software malfunction in your unit and not settings wrong

Endkod, 2024-09-24:
[image]
This is how the screen now looks very standard. All other controls are not present.  Park Assist Display on cockpit does work, which I had set last year

Endkod, 2024-09-24:
USB also works

jinzo, 2024-09-24:
And you did try to 'restore' factory configuration with the data the Telegram pyren bot gives you?