cedricp/ddt4all #1259: Rear camera CLIO V

Kagehoshiii, 2024-09-26:
Hello everyone!

I recently bought a used Clio V, Intens model from April 2021. I had planned to activate some options, including the rear camera, using the latest version of ddt4all, the November 2021 database, and the ecu.zip file provided by Furtif.

However, I am encountering an issue: when I try to write to Navigation/Camera_System_config, the application crashes. Below is the error message and the logs after a scan. The scan indicates that the navigation file is missing. Upon checking the reference 5805_211206, it seems I have a recent ECU, Branch 2.

How can I activate the rear camera? Am I doing it correctly? I have already managed to activate some options via Navigation/Information System.

Thank you in advance for your help.

[image]

[image]

arn-2469, 2024-09-27:
I can't comment on the crashing, but have you bypassed or unlocked the gateway? You won't get far without doing it.

Furtif, 2024-09-27:
is this https://github.com/cedricp/ddt4all/discussions/845 ?????

Kagehoshiii, 2024-09-27:
yes this is it. I got the error as in the screenshot, then I tried again with the ECU.zip file but no change. I checked via the code that the file is loaded correctly

Furtif, 2024-09-27:
> yes this is it. I got the error as in the screenshot, then I tried again with the ECU.zip file but no change. I checked via the code that the file is loaded correctly

humm ok ... Clio V is not blocked by gateway ?

Kagehoshiii, 2024-09-27:
yes, I think so. in any case I was able to activate other options. regarding the rear view camera, for activation, am I in the right place?

Furtif, 2024-09-27:
> yes, I think so. in any case I was able to activate other options. regarding the rear view camera, for activation, am I in the right place?
> 
> Edit : if we connect directly to 2 PIN of the car radio, which is behind the glove box, we bypass the gateway and we can write. from the ODB socket, we can only read. I was able to activate Android auto/CarPlay wirelessly like that. Otherwise I tried with PyRen, in DDT mode, and I was able to write!!!!! except that it didn't change anything, I don't have the reversing camera activated... in I go to the parking assistance, I only have the rear radars.

i go move you problem to https://github.com/cedricp/ddt4all/discussions and please look some issues about clio V and gateway,

if is a bug python ok reopem issue 

### Note:
 - I thinks if this bug happens it raise says (please unlock or byass gateway before) go look in you picture up and implement it, if is possible.

Furtif, 2024-09-27:
Test commit https://github.com/cedricp/ddt4all/commit/9eae438f242ecf7bde51b991e3bdac70613ba7f7

Furtif, 2024-09-27:
Plaese try last commits thanks.

Kagehoshiii, 2024-09-28:
Okay, I'll try today or tomorrow at the latest.
Thanks Furtif

Furtif, 2024-09-30:
> Okay, I'll try today or tomorrow at the latest. Thanks Furtif

tested ????

Kagehoshiii, 2024-09-30:
> > Okay, I'll try today or tomorrow at the latest. Thanks Furtif
> 
> tested ????

I didn't have time this weekend... I might test this tonight.

Kagehoshiii, 2024-10-03:
I tried with the fix but I have another error... finally just for the activation of the camera I used another tool. Thanks for your help!
[image]

Kagehoshiii, 2024-10-07:
> > I tried with the fix but I have another error... finally just for the activation of the camera I used another tool. Thanks for your help! ![IMG_20241001_172554](https://private-user-images.githubusercontent.com/183002657/373392173-d459b8a1-13cb-4e2f-8659-6cd1590887b2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgzMTUzMDAsIm5iZiI6MTcyODMxNTAwMCwicGF0aCI6Ii8xODMwMDI2NTcvMzczMzkyMTczLWQ0NTliOGExLTEzY2ItNGUyZi04NjU5LTZjZDE1OTA4ODdiMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwN1QxNTMwMDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNGEzYzViNjFiMjE4ZTI0MGQ4MWMxYzIzOWEyOWYxZmM5NmE4YmU1NDFiNTNiMDhkMjE3YzdhMzY3ZTllNGM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.m8jBYqOaaUJ7GPdA9mBoeoncgjejDirx5Y1WIq3hkMo)
> 
> message is showed in red in ddt4all window log ?

No, only on VSCode, in the terminal. After that, ddt4all closed. Without VSCode, I only see the error in the CMD terminal but a few seconds. It closed at the same time as ddt4all

Furtif, 2024-10-09:
@cedricp le key e mauvais doit avoir un try catch pareil en ecu.py et paramètres pareil mais j'ai pas se ecu pour test faudra le add en mode test demo peut être tas une idée de quel ecu il utilise ???

Furtif, 2024-11-06:
Hello,
try https://github.com/cedricp/ddt4all/commit/8a1b503762490671b10bff3138076c3b61f6d02b please.
thanks.

Alex-mtn, 2024-11-30:
> I tried with the fix but I have another error... finally just for the activation of the camera I used another tool. Thanks for your help! [...]

Hi! Did you use Pyren? Or another tool? Please let me know how you activated the camera?

Kagehoshiii, 2024-11-30:
> Hi! Did you use Pyren? Or another tool? Please let me know how you activated the camera?

Hi! I used PyRen yes. I bought this adapter on Aliexpress, with the camera. 
[image]
[image]
[image]

Alex-mtn, 2024-11-30:
> > > I tried with the fix but I have another error... finally just for the activation of the camera I used another tool. Thanks for your help! ![IMG_20241001_172554](https://private-user-images.githubusercontent.com/183002657/373392173-d459b8a1-13cb-4e2f-8659-6cd1590887b2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5NjkzMDgsIm5iZiI6MTczMjk2OTAwOCwicGF0aCI6Ii8xODMwMDI2NTcvMzczMzkyMTczLWQ0NTliOGExLTEzY2ItNGUyZi04NjU5LTZjZDE1OTA4ODdiMi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEzMFQxMjE2NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mOWQ2ZmU2MjY1YTA2MzJiYjY1N2I2NDIzY2U2ODNlZDhlM2QzZGM2N2E1NWJhNjc3ZWRiZjA3YjIxNWMwYTNhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QXH3clFir-zy8n1TA3hRwtCQnWUYh_VTmcT-1S1CyXs)
> > 
> > 
> > Hi! Did you use Pyren? Or another tool? Please let me know how you activated the camera?
> 
> Hi! I used PyRen yes. I bought this adapter on Aliexpress, with the camera. ![IMG_20240928_001928](https://private-user-images.githubusercontent.com/183002657/391248081-8e6f29d9-1a5f-494e-ba73-d17761a1ff01.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5ODI1NDcsIm5iZiI6MTczMjk4MjI0NywicGF0aCI6Ii8xODMwMDI2NTcvMzkxMjQ4MDgxLThlNmYyOWQ5LTFhNWYtNDk0ZS1iYTczLWQxNzc2MWExZmYwMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEzMFQxNTU3MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZDc1N2FlOTA2ZDkxYjU3YTNmMjZjZTBkMmVjNzk4MmQ0ODhmY2QzYjUyNDgzMGZlODZlM2VhNGY5ZGE1NWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.PjaYqbiZeySkT6Sy6dwOg0x2X-NT8uWl_W3q4pIwrkk) ![IMG_20240928_094721](https://private-user-images.githubusercontent.com/183002657/391248086-9f3d6223-1fa2-48f2-b2e9-ddd05091e81e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5ODI1NDcsIm5iZiI6MTczMjk4MjI0NywicGF0aCI6Ii8xODMwMDI2NTcvMzkxMjQ4MDg2LTlmM2Q2MjIzLTFmYTItNDhmMi1iMmU5LWRkZDA1MDkxZTgxZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEzMFQxNTU3MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NTU1OWE4YzI5NWM5NmU3ZGUxNWMyZjU4ZGFlNTk1NjUwODJhOGI2NzFmMjI5MTA1ZDlkMzg1NGYxNDZkZjIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.iEjxusRUdmPSlHhLCHmU7gh8pgPTk7JowLG3YcA-TvQ) ![IMG_20240923_181428](https://private-user-images.githubusercontent.com/183002657/391248089-4add655c-8744-4038-817b-767aad52e072.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5ODI1NDcsIm5iZiI6MTczMjk4MjI0NywicGF0aCI6Ii8xODMwMDI2NTcvMzkxMjQ4MDg5LTRhZGQ2NTVjLTg3NDQtNDAzOC04MTdiLTc2N2FhZDUyZTA3Mi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEzMFQxNTU3MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OTg4MGY5ZTE2OGVmODcwNjYyMGZhMjA1ZWY1YTdmMjFhN2UzNDUzODYwMTQ5Nzk5NWM1OWMxYTQ5ZGRiMWFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.je8QXWnpljUEnIKPaZ2weSDinW0NBsMJW4OuUKNtGMQ)

Can you write the activation procedure in Pyren? I will be very grateful

Kagehoshiii, 2024-11-30:
> Can you write the activation procedure in Pyren? I will be very grateful

It was 2 months ago, I don't remember exactly.
I already had to use PyRen because ddt4all crashed when I wanted to write in Navigation/Camera_System_config (maybe not anymore) otherwise for everything else it worked perfectly!
So in addition to PyRen you need a database, I think I had the one from 07.2021, but before that it should work. then you need in Navigation/Camera_System_config, you have to configure like the photo sent just before. there is a last setting to make on the same page, but lower down, to activate the guide lines and that they are not reversed.
I would have to relaunch it to see exactly the manipulation but I will not be able to before Monday

Alex-mtn, 2024-11-30:
> It was 2 months ago, I don't remember exactly. I already had to use PyRen because ddt4all crashed when I wanted to write in Navigation/Camera_System_config (maybe not anymore) otherwise for [...]

I successfully connected wireless carplay via ddt4all, but I can’t configure the camera in Navigation/Camera_System_config. I have successfully connected Pyren with the current database, but I don't know exactly what to do next. I assume that I need to go to the Navigation ECU and change some camera parameter via extended command set.

Kagehoshiii, 2024-11-30:
> I successfully connected wireless carplay via ddt4all, but I can’t configure the camera in Navigation/Camera_System_config. I have successfully connected Pyren with the current database, but I don't [...]

That's right! To activate the rear camera I configured it as in this photo. However, it was difficult, I had to play with several settings so that it finally worked. At times I didn't have the camera (just radars) then the camera area was displayed but without an image. And eventually it worked
[image]

Alex-mtn, 2024-11-30:
> 
> That's right! To activate the rear camera I configured it as in this photo. However, it was difficult, I had to play with several settings so that it finally worked. At times I didn't have the camera (just radars) then the camera area was displayed but without an image. And eventually it worked !

But this photo is from ddt4all, in ddt4all I get the message "please unlock or bypass gateway before" and I can't do it. Do you do this in ddt4all? I'm totally lost

Furtif, 2024-11-30:
> But this photo is from ddt4all, in ddt4all I get the message "please unlock or bypass gateway before" and I can't do it. Do you do this in ddt4all? I'm totally lost

is last commits ?

Alex-mtn, 2024-11-30:
> is last commits ?

yes

Kagehoshiii, 2024-11-30:
> But this photo is from ddt4all, in ddt4all I get the message "please unlock or bypass gateway before" and I can't do it. Do you do this in ddt4all? I'm totally lost

From PyRen you have the same view

Alex-mtn, 2024-11-30:
> yes

[image]

Alex-mtn, 2024-11-30:
> From PyRen you have the same view

wow, I thought in pyren we only work in the terminal. This is DDT mode?

Furtif, 2024-11-30:
> > > > > That's right! To activate the rear camera I configured it as in this photo. However, it was difficult, I had to play with several settings so that it finally worked. At times I didn't have the camera (just radars) then the camera area was displayed but without an image. And eventually it worked !
> > > > 
> > > > 
> > > > But this photo is from ddt4all, in ddt4all I get the message "please unlock or bypass gateway before" and I can't do it. Do you do this in ddt4all? I'm totally lost
> > > 
> > > 
> > > is last commits ?
> > 
> > 
> > yes
> 
> <img alt="2024-11-30 в 20 17 53" width="1126" src="https://private-user-images.githubusercontent.com/190185687/391272405-0132bd85-78b4-45ac-9e26-e0a1e92f9d35.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5OTU0NzUsIm5iZiI6MTczMjk5NTE3NSwicGF0aCI6Ii8xOTAxODU2ODcvMzkxMjcyNDA1LTAxMzJiZDg1LTc4YjQtNDVhYy05ZTI2LWUwYTFlOTJmOWQzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEzMFQxOTMyNTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYTMxMWQyMzc4OTNlMWZlODhiYjY2NWU2MDBlOTg1MGU2YWQxMDI1MDU0NTc3NzljYjZjOGNjNGQzMzZmMzcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.3ZT6cKr1qgSOGjfoOxqm48aBL4q9rVhEXImrBnKsVUE">

seen ok before it get error

Furtif, 2024-11-30:
is the zoe electric or clio v and more recent models need baypass Gateway

Alex-mtn, 2024-11-30:
> is the zoe electric or clio v and more recent models need baypass Gateway

yes, this is a 2020 Zoe with CGW. I connected directly to the navigation ECU and have already successfully changed many parameters... except the camera