cedricp/ddt4all #1290: ddt4all crash on writing tuning

cleb40, 2024-11-05:
car : renault megane 2 phase 2

Hello,
 i just tried to write the tuning of electric power steering
the ECU file is DAE_X84_Serie_V1.json > diagnostics > select tuning
i activated expert mode
ddt4all crashed when pushing the button to write the new tunning
the logs and crash errors are in the joined archive.
something about expecting string but got integer instead
[crash_logs.zip](https://github.com/user-attachments/files/17636167/crash_logs.zip)

cleb40, 2024-11-06:
Ok, after loads of debugging i found out these lines in ecu.py in method Ecu_data.setValue():
646        # TODO this need review, return empty is not maybe ok.
647        bytes_list = {}

it seems line 647 needs to be commented out.
i explain what i found : 
When i call to write tuning the elm_data_stream built by this function is an array with an unique index of 2 and value of '01' ( for tunning number 1 )
This array crash the code : ' '.join(elm_data_stream) , dont ask me why i'm no python expert

However since i'm getting the hang of how things work : 
For writing tunning the "sentbytes" is  "3B0200" , with the last 2 char ( last byte )  being the tuning number.
This last Byte must be updated with the tunning value wich is what the method setValue does
However with the line 648 , it reset the whole command to an empty array ( actually a hash ) and a bit further the code  only put the modified 3rd byte back in.

I don't know anything about CAN either , but i guess the whole sentbytes command has to be sent modified with the right tuning and with a space between each byte like this :  '3B 02 01' , which is acheived by commenting out line 647.

I leave the review of this to the experts, and wait for validation before trying anything with my car :)

Thanks

Furtif, 2024-11-06:
https://github.com/cedricp/ddt4all/discussions/1289

closed 2024-11-06 by Furtif