cedricp/ddt4all #1320: Unlock android app

pitarenal, 2024-12-15:
Hello Cedeic,

Frist of all, thanks you for your excellent work.

I need to restore my renault audio system and I need to unlock the android app.

I have read I need to write you to ulock it. Can you help me?

Thank you so much

AndyIreland, 2024-12-15:
android app no longer needs unlocking (not the latest version anyway)

pitarenal, 2024-12-15:
Ok, maybe the problem is my ELM device. I will try with another one.

Thank you!

closed 2024-12-15 by pitarenal

mishu28nmv, 2024-12-19:
> Hello Cedeic,
> 
> Frist of all, thanks you for your excellent work.
> 
> I need to restore my renault audio system and I need to unlock the android app.
> 
> I have read I need to write you to ulock it. Can you help me?
> 
> Thank you so much

Hi,
is there an Android App for this wonderful software? Can I have a link please?

manio, 2024-12-19:
> is there an Android App for this wonderful software? Can I have a link please?

https://github.com/cedricp/ecutweaker

mishu28nmv, 2024-12-19:
Thank you for the link. I will surely try this too!!!