cedricp/ddt4all #1339: D4F BVR NO IN DATABASE DDT4ALL

azuwanautogas, 2025-01-02:
ddt4all : 1. no have data ecu ver D4F BVR
              2. i try to using single name BVR-X06-D4F still cant
              3. can i change number
 but i dont know what version ecu this :``<AutoIdent DiagVersion="8" Supplier="174" Soft="5201" 
                  Version="420E"/></AutoIdents>
              4. can everyone help me how can do 

    thanks for helping

[image]

AndyIreland, 2025-01-02:
use the magnifying glass in DDT4ALL to find ECU's of your car (Choose CAN)

azuwanautogas, 2025-01-02:
Im ready do this still can't.i think need someone modified ecu for me recommendation

< the magnifying glass in DDT4ALL to find ECU's of your car (Choose CAN)