cedricp/ddt4all #1352: [BUG] text not visible in kde linux dark mode

Vincenzoferrara, 2025-01-15:
dark mode in linux kde, but not in app
[image]

if you enable dark mode in the app you get that now everything is visible, but if you open a combox you can't read the content (I couldn't take a screenshot)

Furtif, 2025-01-16:
You Python—version ?

Vincenzoferrara, 2025-01-16:
Python 3.12.7

Furtif, 2025-01-16:
> Python 3.12.7

normaly is good.

Furtif, 2025-01-16:
It's very strange to only have you with this issue, I tested it on Debian with Plasma 6 Macos and Windows and everything was fine.

Furtif, 2025-01-16:
you have try : ``` sudo apt-get install --reinstall libxcb-xinerama0 ```

Vincenzoferrara, 2025-01-16:
I use manjaro (arch), theme how use for kde is: breath dark

Furtif, 2025-02-22:
tested un unbunto desktop theme dark and DDt4ALL dark and yep as issues, need maybe edit https://github.com/cedricp/ddt4all/blob/master/ddt4all_data/qstyle-d.qss or create one for uses if linux ubuntu + desktop theme dark

Furtif, 2025-02-22:
maybe create a modified qstyle-d-lu.qss from base https://github.com/cedricp/ddt4all/blob/master/ddt4all_data/qstyle-d.qss and in: https://github.com/cedricp/ddt4all/blob/master/main.py#L1190 add:

```python
# Need maybe get if is Ubuntu and desktop theme is Dark
if sys.platform[:3] == "lin":
     stylefile = core.QFile("ddt4all_data/qstyle-d-lu.qss")
else:
     stylefile = core.QFile("ddt4all_data/qstyle-d.qss")
```

Furtif, 2025-02-22:
@Vincenzoferrara in your KDE as normal (Not Dark mode) runs good ?

Furtif, 2025-02-22:
https://github.com/cedricp/ddt4all/commit/0525d920319c68545698f1befcb0e40b3f683430

Furtif, 2025-03-06:
try last commits please

closed 2025-03-10 by Furtif