cedricp/ddt4all #1374: Different options in ddt4all and ecu-tweaker

pczolee, 2025-02-22:
Hi all.

First of all, thanks for this great programs. I usually using ecu tweaker, because it is easier from a smartphone. But I faced a problem. In X10 Zoe, with ddt4all there is a lot more options to setup the rear camera guidelines than in the ecu tweaker. I'm using the same database for both.

Here is a screenshoot from ddt4all:

[image]

And here is from Ecu-Tweaker:

[image]
[image]

Thanks if you can help me.

Furtif, 2025-02-22:
Is last tweak ? https://github.com/cedricp/ecutweaker/releases/tag/v2.4 and is same DB ?

Furtif, 2025-02-22:
is fine 

[image]

Update your apps please

closed 2025-02-22 by Furtif

pczolee, 2025-02-22:
Well, with ddt4all is fine (you can see on my screenshot), but not with ecu-tweak. Of course I tried different versions, no luck :(

Furtif, 2025-02-22:
uses https://github.com/cedricp/ddt4all/releases/tag/v3.0.3 and https://github.com/cedricp/ecutweaker/releases/tag/v2.4 and DB ecu.zip https://github.com/cedricp/ddt4all/discussions/845 for both.

Furtif, 2025-02-22:
in your screen DTT4ALL i thinks you uses Linux Ubuntu, reclone repos fallow readme file.

Furtif, 2025-02-22:
or un clone folder type `git pull` .....

pczolee, 2025-02-22:
The problem is with ecu-tweak!!! In the ddt4all everything is fine.. I have the same db.

Furtif, 2025-02-22:
NO the problem is you old version DDT4ALL make note last uses all fix for all address of DB

pczolee, 2025-02-22:
Well, I have new ddt4all as well, but there is no difference regarding this. So,just to make clear, the problem is NOT with ddt4all! The problem is with ecu-tweaker (android)!

Furtif, 2025-02-22:
update also https://github.com/cedricp/ecutweaker/releases/tag/v2.4 and put ecu.zip in both (linux ddt4all) and in android

Furtif, 2025-02-22:
Or you not uses DDT4all  ok ... tested 3 times as x10 ZOE renault and fine

Furtif, 2025-02-22:
DELETE DDT2000 normal DB from DDT4ALL clone or source uses only ecu.zip

Furtif, 2025-02-22:
[image]

Furtif, 2025-02-22:
Linux ubuntu 2x.xx python 3.12.x

Furtif, 2025-02-22:
ecu-tweaker (android) is ok also ... but have some issues because new permitions android (recent releases 12-13-14-15)

pczolee, 2025-02-22:
Show me the same display is ecu-tweaker. I have missing buttons (you can see my screenshoots).

Furtif, 2025-02-23:
Hum yep in android miss some button

Furtif, 2025-02-23:
Java parser maybe but ok need look but is side https://github.com/cedricp/ecutweaker not ddt4all

pczolee, 2025-02-23:
Yes, now we are speaking about the same thing ;) . Thanks for your attention. 
But because the same developer I thought it is ok to talk about this here. Should I open the issuie in ecutweaker site?

Furtif, 2025-02-23:
> Yes, now we are speaking about the same thing ;) . Thanks for your attention. But because the same developer I thought it is ok to talk about this here. Should I open the issuie in ecutweaker site?

yes move https://github.com/cedricp/ddt4all/issues/1374 to https://github.com/cedricp/ecutweaker/issues