cedricp/ddt4all #1375: Nissan Navara not showin up in ECU Tweaker list of cars

AndyIreland, 2025-02-22:
hi - can find other Nissan's in the list of cars and I can find Navarra X60A in DDT4ALL but for some reason I cannot find it in ECU Tweaker for Android - I am using the ECU.ZIP from DDT4ALL.

Thanks.

@cedricp  & others

[image]

Furtif, 2025-02-23:
And is in [image]

Furtif, 2025-02-23:
https://github.com/cedricp/ecutweaker/releases/tag/v2.4

AndyIreland, 2025-02-23:
brillant thak you @Furtif  - I will try it

Furtif, 2025-02-23:
> brillant thak you [@Furtif](https://github.com/Furtif) - I will try it

ok close this please.

closed 2025-02-23 by AndyIreland