cedricp/ddt4all #1377: Wifi Dongle : unable to connect wifi device (Renault Zoé 2015)

Jodaille, 2025-02-24:
Hello,

I have a Wifi ELM with pic18F25k80 4Mhz

I have tried to connect with DDT4ALL (Linux) and ECU Tweaker (v2 debug) without success.

I can connect with EODB Facile
With it, I can remove error codes, but I have water pump P0463-F1 that reappear.

It seems we need to reset pump counter 

https://renault-zoe.forumpro.fr/t19006p50-verification-des-compteurs-de-la-pompe-a-eau-et-si-necessaire-remise-a-zero
2E334900000000
2E334A00000000
2E334B00000000
2E353100000000

I can read python a bit (linux)

What could I do to debug the connection process ?

[image]

Furtif, 2025-02-24:
sudo ufw allow 35000

Or intall firewall gui
sudo apr-get install hufw and config firewall Linux.

closed 2025-02-24 by Furtif

Furtif, 2025-02-26:
Thanks 👍, need maybe add a stuff or option.

Jodaille, 2025-02-26:
> Thanks 👍, need maybe add a stuff or option.

Something like that ?
[image]

Furtif, 2025-02-26:
> > Thanks 👍, need maybe add a stuff or option.
> 
> Something like that ? [image]

yep seen cool make a PR

Jodaille, 2025-02-26:
I have made a PR, let me know if it is ok for you.
I would like to check  "in real conditions" tomorrow

Furtif, 2025-02-26:
Ok go look 👀 thanks

Jodaille, 2025-02-27:
I have test the changes in the car, it seems to work.
If I check it connects.

For memory :
If not checked it "hangs" on 
Trying to open port 192.168.0.10:35000 @ 38400

If I Ctrl-c I receive plenty of:
[Errno 9] Bad file descriptor

and then again : 
Trying to open port 192.168.0.10:35000 @ 38400

Furtif, 2025-02-27:
yep i have look but add # in lines print (or debug logs) not need and make unused strings for translations

Jodaille, 2025-02-27:
Sorry for the print, I forgot to remove.

do you want to remove translated string ?
`sockettimeoutlabel.setToolTip(_("Enable socket timeout"))`

Furtif, 2025-02-27:
remove or comment unneded print log

Jodaille, 2025-02-27:
Ok, I have removed print log and tooltip

Furtif, 2025-02-27:
> Ok, I have removed print log and tooltip

bad still look https://github.com/cedricp/ddt4all/pull/1379

Jodaille, 2025-02-27:
Well, sorry I could not see what I could change, is it the function name ?

I have add gettaddr on options in last commit :
https://github.com/cedricp/ddt4all/pull/1379/commits/bcc19995ad9765028357eb3f02f9e06bc5445f53

Could you please give me a pointer :)

Furtif, 2025-02-27:
> Well, sorry I could not see what I could change, is it the function name ?
> 
> I have add gettaddr on options in last commit : [bcc1999](https://github.com/cedricp/ddt4all/commit/bcc19995ad9765028357eb3f02f9e06bc5445f53)
> 
> Could you please give me a pointer :)

add var in options.py

https://github.com/cedricp/ddt4all/blob/master/options.py#L70 is loaded here

Jodaille, 2025-02-27:
Nice, I have searched where options was set, now I know thank you.

I have pushed.

commit 4fdb7f4f (2025-02-27): Merge pull request #1379 from Jodaille/master

add checkbox to enable socket settimeout with wifi
fix  #659
fix  #1377