cedricp/ddt4all #1380: New plugin how to write value for reseting counter

Jodaille, 2025-03-02:
Hello,

I try to write a new plugin for reseting water pump counters (low, middle, high speed).

I have look at the other plugins, I can read values but resting (write) return None.

For example: 
https://github.com/Jodaille/ddtplugins/blob/master/zoe_waterpump_counter_reset.py#L216-L224

`        reset_request = self.evc_ecu.requests[u"DataWrite.($3349) Time Counter for the driving WEP in Low Speed"]
        request_response = reset_request.send_request()`

Nb: I start diag session before https://github.com/Jodaille/ddtplugins/blob/master/zoe_waterpump_counter_reset.py#L174-L181

Can you see what am I missing ?

Jodaille, 2025-03-02:
In xml:
EVC_3180_RH5_510_V1.1_20210422T184714.xml

I see StartDiagnosticSession.Default (I use this one to start session)
but there is two others:

StartDiagnosticSession.Programming
and
StartDiagnosticSession.ExtendedDiagnosticSession

Do you think I have to use this one to be able to write (reset) counter ?

Jodaille, 2025-03-03:
Hello,

I may have found something, thanks to a video: https://youtu.be/cNc5uVA0Jf8?t=511
I need to pass 0 as value, and it seems it correspond to 
<DataItem Name="($3349) Time Counter for the driving WEP in Low Speed" in XML file

`
        reset_request = self.evc_ecu.requests[u"DataWrite.($3349) Time Counter for the driving WEP in Low Speed"]
        request_response = reset_request.send_request({"($3349) Time Counter for the driving WEP in Low Speed": 0})
`

Jodaille, 2025-03-04:
Hello,

I can reset low speed counter using start_diag_extend_session()
\o/
I have changed medium, high and V_Timer_DrivWEP_ON to use it (but no tested now)

Furtif, 2025-03-05:
I not have access in repo plugins this need @cedricp

Furtif, 2025-03-11:
[image]

need verif

Furtif, 2025-03-11:
make TIPS as translatable , fix screen please.

Jodaille, 2025-03-11:
> make TIPS as translatable , fix screen please.

About screen I am using:
` self.setGeometry(100, 100, 480, 445)`
https://github.com/Jodaille/ddtplugins/blob/master/zoe_waterpump_counter_reset.py#L26C9-L26C45
That seems to work for me.

[image]

Do you know a best way to do it ?
Or do you know why it does not work for you ?

Furtif, 2025-03-12:
https://github.com/cedricp/ddt4all/blob/master/ddtplugins%2Fzoe_waterpump_counter_reset.py

Jodaille, 2025-03-12:
Hello,

I have pushed "Tips translate"

I have added self.setFixedSize(480, 480) 
but I cannot not test it on Windows
If you can have a try 

Thank you
https://github.com/cedricp/ddt4all/pull/1388

Jodaille, 2025-03-13:
Hello !

please see PR
https://github.com/cedricp/ddt4all/pull/1391

closed 2025-03-13 by Furtif