cedricp/ddt4all #1381: renault clio (c1a) connection problem

eemrecakar, 2025-03-04:
Hello, I have a clio mk5 vehicle. I downloaded your current version program, 3.0.3, but when I select my vehicle and press the magnifying glass, the application suddenly closes.

I tried your older versions, 3.0.2, I had no problems, I even activated carplay, but this time when I wanted to activate my rear camera, the application suddenly closes again.

https://github.com/user-attachments/assets/660f156f-d91f-42fa-949c-dccf35492a73

Jodaille, 2025-03-04:
Hello,

can you try to run the program in a terminal to show the error ?
I don't know windows anymore but you have the command in the Icon properties.

Furtif, 2025-03-04:
> Hello, I have a clio mk5 vehicle. I downloaded your current version program, 3.0.3, but when I select my vehicle and press the magnifying glass, the application suddenly closes.
>
> I tried your older [...]

Change ddt4all_data/projects.json uses new in repos

Furtif, 2025-03-04:
https://github.com/cedricp/ddt4all/blob/master/ddt4all_data%2Fprojects.json

eemrecakar, 2025-03-04:
Hello, thank you very much for your support, my problem has been solved thanks to you.

Furtif, 2025-03-04:
> Hello, thank you very much for your support, my problem has been solved thanks to you.

Ok 👍 cool 😎 close this please.

closed 2025-03-04 by eemrecakar