cedricp/ddt4all #1382: Dark display without dark mode on Mac OS

Jodaille, 2025-03-04:
Hello,

on Mac OS, I have a strahnge display (without Dark mode enabled)

[image]

If I add :

```
QWidget {
    color: #000000;
    background-color: #ffffff;
}
```
in ddt4all_data/qstyle.qss

[image]

is it the proper method, the right color ?

Furtif, 2025-03-04:
No

Furtif, 2025-03-04:
Normaly mac os is ok , https://github.com/cedricp/ddt4all/issues/1352

Furtif, 2025-03-04:
maybe combobox bad in last python

Furtif, 2025-03-04:
[image]

Jodaille, 2025-03-04:
The qss file loaded is:
ddt4all_data/qstyle.qss

the first image is a screenshot on Mac OS, repository just cloned today. if it helps.

Furtif, 2025-03-04:
> The qss file loaded is: ddt4all_data/qstyle.qss
> 
> the first image is a screenshot on Mac OS, repository just cloned today. if it helps.

no is qstyle_d.qss

Furtif, 2025-03-04:
file in windows seem ok i have create new for linux kde but need fix maybe edit and uses both for linux kde / mac

Jodaille, 2025-03-04:
> no is qstyle_d.qss

[image]

I have "dark": false, in config.json

Furtif, 2025-03-04:
> > no is qstyle_d.qss
> 
> [image]
> 
> I have "dark": false, in config.json

if false dark is disabled and uses normal qss

Furtif, 2025-03-04:
normal seem ok

[image]

Jodaille, 2025-03-04:
OK, strange

closed 2025-03-04 by Jodaille

Jodaille, 2025-03-04:
It works well on another mac 🤷🏻‍♂️

Furtif, 2025-03-04:
> It works well on another mac 🤷🏻‍♂️

Yep Depends Python version I thinks , I m in last macOS (not beta)

Furtif, 2025-03-04:
join discord https://discord.gg/yAZvqDuj

Furtif, 2025-03-05:
or delete all database and unses zip https://drive.google.com/file/d/14AQTXRv4Y0mVnWNeZ6seznKPRd-7ZaKm/view?usp=drive_link

Furtif, 2025-03-05:
seem same ... need a review

Jodaille, 2025-03-05:
> seem same ... need a review

Can you reproduce it ?

Jodaille, 2025-03-05:
Same versions of packages between two machines:
crcmod==1.7
PyQt5==5.15.11
PyQt5-Qt5==5.15.16
PyQt5-stubs==5.15.6.0
PyQt5_sip==12.17.0
PyQtWebEngine==5.15.7
PyQtWebEngine-Qt5==5.15.16
pyserial==3.5
pyusb==1.2.1

One with python 3.11.5 (works) the other 3.11.6

It could be Mac OS version: 
Mac OS ventura ("bad display")
Mac OS big sur (ok)

Furtif, 2025-03-05:
> > seem same ... need a review
> 
> Can you reproduce it ? 

For now not.
Need maybe change conditional.

Furtif, 2025-03-05:
> Same versions of packages between two machines:
> crcmod==1.7
> PyQt5==5.15.11
> PyQt5-Qt5==5.15.16
> PyQt5-stubs==5.15.6.0
> PyQt5_sip==12.17.0
> PyQtWebEngine==5.15.7
> PyQtWebEngine-Qt5==5.15.16
> pyserial==3.5 [...]

Mine is sequoia

Furtif, 2025-03-05:
[image]

test

Jodaille, 2025-03-05:
OK in Dark mode for me to

[image]

Furtif, 2025-03-05:
cool