cedricp/ddt4all #1386: Wiki outdated versions links EcuTweaker

Jodaille, 2025-03-07:
Hello,

https://github.com/cedricp/ddt4all/wiki/Android-port

V2.0 does not work for me ("Unable to connect wifi device")
[image]

The last release is 2.4 (works/connect)
https://github.com/cedricp/ecutweaker/releases/tag/v2.4
but not indicated on the port page

[image]

I have not found how to contribute to wiki page (could not checkout)

Furtif, 2025-03-07:
yep but now for releases join releases in ecutweaker repo

Furtif, 2025-03-07:
- GitHub releases (uses debug build)

Jodaille, 2025-03-07:
Hello,

sorry my message/issue is not clear.

The last release 2.4 works,
but V2.0 is mentioned on ddt4all Wiki page:
https://github.com/cedricp/ddt4all/wiki/Android-port

Furtif, 2025-03-07:
> Hello,
> 
> sorry my message/issue is not clear.
> 
> The last release 2.4 works, but V2.0 is mentioned on ddt4all Wiki page: https://github.com/cedricp/ddt4all/wiki/Android-port

ok fixed, setted 2,4

Jodaille, 2025-03-07:
Great, thank you !

closed 2025-03-07 by Jodaille