cedricp/ddt4all #1446: Tom tom to rlink Megane 3

kajoQ, 2025-05-04:
Hello,

I need help to do a config rlink im Megane 3 from tom tom without live. I have issue with radio sterringwheel control; when i'm scrolling radiostatio,it change channel from1,2,4,2,5, not like 1,2,3,4,5 ... and radio sometimes can frozen and i must turn car off or take off radio supply off.
Instalation and other radio,screen etc. arę ok.
Probably i need do do a radio config on screen navigation. I do a can2 plug and connect więtcj ddt4all but  in navigation when a choose mfd 5.4 or nfa 0.6 or 2.5 the parameters is no data.  
Maybe i choose wrong ecu? Please some help.

AndyIreland, 2025-05-04:
if it is definitely rlink1 and not rlink2 - what Version number is your software on it?

If you have definitely wired up the splitter cable or ELM327 for can2 and if your ELM327 has definitely got a PIC18F25K80 chip in it and not a cheep chinese one, change CAN timeout in DDT4ALL to 600ms and you might need to change this SWRC parameter:

[image]

kajoQ, 2025-05-05:
Its rlink 1,soft 11.347. I have elm bluetooth vgate2  but i have elm on cable top (  seller write that is on this PIC). SWRC change on what type must be ? That timeout ddt4all  where can i change ? Can i do it in ECUwteaker or only ddt4all?