cedricp/ddt4all #1449: activation Android auto sur grand scenic 3 phase 2

zorq512, 2025-05-08:
Bonjour,

Je cherche à activer android auto sur le R-link1 de mon Grand scenic 3 phase 2. Ca fait des semaines que je fais des recherches sans arriver à mes fins alors je viens vers vous pour soliciter votre aide.

J'ai suivi  les instructions mais je suis apparement passé à coté de quelque chose

- j'ai acheté un ODB2 wifi elm327
- j'ai modifié une ralonge pour matcher les recommandations du post [https://blog.zwindler.fr/2022/04/08/androidauto-rlink1-cherie-jai-brique-la-voiture/](url)
- j'ai téléchargé DDT4all All cars sur le site russe [https://www.drive2.com/b/498093336985338243/](url) comme indiqué dans ce post [https://github.com/cedricp/ddt4all/issues/614](url)

Quand je branche tout ca, et que je suis les indications toutes les valeures sur la partie droite sont en rouge comme le montre la capture ci dessous

[image]

Je n'ai pas trouvé de database ddt2000 est ce indispensable?
Dans la liste des vehicules je n'ai trouvé que scenic 3 phase 2 mais de "**Grand** scénic" est ce le meme model pou DDT4all?

Merci d'avance pour aide

AndyIreland, 2025-05-08:
try changing the CAN timeout to 600ms and refresh. - has your ELM327 got the required PIC18F25K80 chip in it or is it a cheap chinese clone?

zorq512, 2025-05-08:
Hi Andy,

thx for your help, I will try to change the time out value.

for the ELM, I bought this one on amazon [https://www.amazon.fr/dp/B0CXW88FFM](url) I think it is a chinese one

zorq512, 2025-05-08:
Hi change the refresh rate and the time out to 600ms, the result is the same
[image]

in the python console I receive d the message "QLayout Cannot add null widget to QHBoxLayout/"
[image]

 I try why a second ODB2 device : a kungfuren which look like to be also a chinese model and the result is also the same

AndyIreland, 2025-05-08:
you sure its a rlink1 in your scenic? - I thought they had rlink2 unit. what version number does it say in the radio's menu?

zorq512, 2025-05-08:
Yes it is a r-link 1, this is confirmed by myrenault web site 

[image]

the version should be the last one as I managed all the update recently. I will confirmed the version number ASAP

zorq512, 2025-05-08:
material ID : R-link TomTom live
the version is 11.346.3206684-8060

zorq512, 2025-05-08:
yes, I have modified the cable extension as shown.

when I run the test all finish with an "OK" statement except v1.4 ATFI which fail

zorq512, 2025-05-09:
I manage several test and research to try to find the problem.

I found this link where people reach the solution [https://megane3.fr/autoradio-gps/topic18827-120.html](url).

I change my cable spliter, see below how it is wired, can you cofirmed that am I'm right

[image]

when I activate the "log all" I see all request receive in blue until I activate expert mod and after all request are receive in red, when I stop expert mod request restart to be receive in bleu

[image]

I try to change the value from "not preset" to "present" for 3 field (SPVR iPhone, Android auto feature and mirorlink feature) this action is in the log.

do you see something wrong?

zorq512, 2025-05-09:
I have find the solution. Th cable is right now with the configuration above, I have download the last version of DDT4ALL, reuse the ECU.zip file and connect the Bluetooth ELM (the wifi one don't work. all the data where collect and the right side is not red anymore.
Android auto work well. thanks Andy for your help and many thanks Cedric for this great tool!

closed 2025-05-09 by zorq512