cedricp/ddt4all #1468: Could not initialize GLX

kleyton67, 2025-06-02:
Steps to reproduce
1. Download master branch
2. git submodules update --init
3. conda create -n ddt4all python=3.10
4. conda activate ddt4all
5. python main.py

log
```bash
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
Abortado (imagem do núcleo gravada)

```

OS: 
cat /etc/issue
Ubuntu 25.04 \n \l

IF you need mre info let me know.

Furtif, 2025-06-03:
> Steps to reproduce
> 
> 1. Download master branch
> 2. git submodules update --init
> 3. conda create -n ddt4all python=3.10
> 4. conda activate ddt4all
> 5. python main.py
> 
> log
> 
> qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
> qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
> Could not initialize GLX
> Abortado (imagem do núcleo gravada)
> OS: cat /etc/issue Ubuntu 25.04 \n \l
> 
> IF you need mre info let me know.

You need maybe runs pip install-r requirements.txt

https://github.com/cedricp/ddt4all#how-to-install-the-application-

closed 2025-06-03 by Furtif

kleyton67, 2025-06-03:
Sorry, I forgot, i did it now, and it didn't worked. I think the card must not be closed.